google/cro3 #22: Add a command to get the current synced version from cros-sdk directory

hikalium, 2023-06-26:
We can query the synced version by portageq-${board} envvar PORTAGE_BINHOST.

This is good for synced up the DUT with currently checked out cros-sdk.

HidenoriKobayashi, 2023-07-04:
I'd like to try this one, but I need clarifications.

The following is an example output for brya on my local workstation
```
 gs://chromeos-prebuilt/board/amd64-generic/postsubmit-R116-15505.0.0-52093-8778090689467554065/packages gs://chromeos-prebuilt/board/amd64-generic/postsubmit-R116-15505.0.0-52094-8778089119028579249/packages gs://chromeos-prebuilt/board/amd64-generic/postsubmit-R116-15505.0.0-52095-8778086878663453457/packages gs://chromeos-prebuilt/board/amd64-generic/postsubmit-R116-15505.0.0-52096-8778085176293458977/packages gs://chromeos-prebuilt/board/brya/postsubmit-R116-15505.0.0-83679-8778090996123150033/packages gs://chromeos-prebuilt/board/brya/postsubmit-R116-15505.0.0-83680-8778090645377985841/packages gs://chromeos-prebuilt/board/brya/postsubmit-R116-15505.0.0-83681-8778089074903851649/packages gs://chromeos-prebuilt/board/brya/postsubmit-R116-15505.0.0-83682-8778086834597103329/packages gs://chromeos-prebuilt/board/amd64-generic/cq-R116-15502.0.0-52012-8778246588928878481/packages
```
- Can I just pick the version string (R116-15505.0.0) from the first entry?
- How does the CLI look like? (under `sync`? or a new command like `repo`?)

hikalium, 2023-12-19:
@mhiramat I found that I filed the bug already, so please close this once the implementation is done!!!!