google/cro3 #30: show repo sync progress while `lium sync`

hikalium, 2023-06-26:
Not showing progress on lium sync is not user friendly. display some progress to make it clear that it is not stuck but it is in progress...

brandonpollack23, 2023-10-05:
https://docs.rs/indicatif/latest/indicatif/ might help with that, along with https://crates.io/crates/tracing-indicatif to integrate it with the logging work in #88

brandonpollack23, 2023-10-06:
repo sync already outputs a percentage and overwrites it's line using [console control characters](https://man7.org/linux/man-pages/man4/console_codes.4.html), so all we need to do is read the std out of the process, cut out the percentage (using regex or simple string splitting) and turn that into an int as an input for indicatif.  

Then integrate indacatif with the tracing system with the tracing-indicatif crate.

hikalium, 2023-11-15:
Now it is implemented by @4n12i so closing this as Fixed. Thank you!

closed 2023-11-15 by hikalium