google/cro3 #171: Some DUT info is missing on a VM so `lium dut info` / `lium dut list --add` fails

hikalium, 2023-11-27:
`BOARD=betty` does not have a valid `hwid` so it fails to be added by `lium dut list --add` command.

It looks like `crossystem inside_vm` returns whether the instance is a VM or not, so we can use that flag to use a workaround, like using `betty_${SOME_SERIAL_NUMBER_ISH_THING}` as a dut_id.

Potential sources that can be used for such serial-number-ish strings:

- Network address used for ssh (e.g. `localhost:1234`)
- The host public key of an SSH server of a DUT (ssh-keyscan command should help, but @cynthia reported that it did not work against betty for some reasons )
- Disk UUID? (not sure if it is random per each VM instances though...)

hikalium, 2023-11-27:
@4n12i will take a look at this (in their free time), based on the offline discussion

hikalium, 2023-11-27:
```
ssh-keyscan -v -p 9222 -t dsa,rsa,ecdsa localhost
```
for the host key approach, this is the workaround ( Thanks @cynthia !!! )

4n12i, 2023-12-01:
It seems to work fine if we specify `127.0.0.1` instead of `localhost` like:

```
ssh-keyscan -p 9222 -t dsa,rsa,ecdsa 127.0.0.1
```

(This is probably because the host name in `.ssh./config` is set to `127.0.0.1` by `./betty.sh setup`.)

4n12i, 2023-12-01:
https://docs.rs/openssh-keys/0.6.2/openssh_keys/

closed 2023-12-15 by hikalium